hammockman/breather #4: Design concept: unknown

racleave, 2020-03-24:
@hammockman, can you come up with a concept, quite different to #3, and fill in here? Would ideally like to choose between them on 26th.

hammockman, 2020-03-24:
Some really loose concept fragments:

* share air sources/sinks between many ventilators rather than one per patient
* reciprocating beam pumps
* treadle sewing machines
* super-plentiful, less-than-perfectly effective better than opposite
* air tunnel - rotating patients - no masks

hammockman, 2020-03-24:
Righto, #4a (no idea how different from #3 this is as #3 description sketchy):

* thin bag pairs with volume being increased/decreased equally powered by choice of electric motor, human power
* providing high-p and low-p rails
* patient mask fed with flapper valve at Y centre, +ve actuation on exhalation side only
* possibly shared between more than one patient
* breathing rate controlled by red/infrared oximeter
* no allowance for o2 enrichment
* masks,

racleave, 2020-03-24:
Cheers. Does the diagram in #3 need to be clearer?

As I understand these ventilators we are talking forced ventilation, ie with a tracheostomy tube. I guess masks might be an option for early treatment, but the main focus is a tube down the, er, gullet.

The requirement sheet stipulates that setting of FiO2 is a must, so we need O2 enrichment.

Are these (infra)red sensors cheap?

Where does the new air come in?

racleave, 2020-03-24:
Further to the mask/tube bit, the latest FAQ has a good set of links, including a tech guide on Covid-19. FAQ is here:

https://www.agorize.com/en/challenges/code-life-challenge/pages/faq?lang=en

racleave, 2020-03-24:
Whoops, looks like my O2 enrichment comment above is wrong - the new requirements sheet says 

8. Oxygen concentration NOT mandatory, recommended

but also that

2. FiO2 from 20% to 100% in discreet steps of 10% 

I obviously need to understand FiO2 a bit better.

hammockman, 2020-03-24:
I read those two bits of info as "O2 enrich not mandatory" but if you do "allow for 10% steping from 20 to 100%"

hammockman, 2020-03-28:
Turns out more knowledgeable ppl have been looking at multi-patient ventilation:

* https://emcrit.org/pulmcrit/split-ventilators/