hammockman/breather #7: Controller software

racleave, 2020-03-25:
This is an issue to deal with design of the controller software, and which hardware it has to interface to. Specific bugs should have their own issues.

We are using a Raspberry Pi 3B+ with a separate LCD screen.

The software has to read various sensors and control 2-6 valves. It also has to have at least two major modes:

 1. Timed breathing
 2. Patient triggered breathing, with a timed override

There are many variations on these modes, but keeping things simple at this stage is a must.

Here are some  is a plot of typical pressures and times (lots more on the web):

[image]
[image]
[image]

racleave, 2020-03-25:
I will set up the raspberry pi here and see how to get @hammockman logged into it. Any preferred OS? I'll go with Raspbian if not.

hammockman, 2020-03-25:
raspbian fine