hammockman/breather #8: Joseph's 5 concepts

racleave, 2020-03-26:
Just saw Joseph's 5 concepts. 

1. This is what we have in issue #3. It seems mainly good to me although I think we could dispense with flow and O2 meters on the exhaust side...and maybe pressure? We should have the bits for this once deliveries come through

2. I think a downside of this is that the common line with the pressure sensors will risk the patient redrawing their last breath - although if there is a flap valve at the Y-connector this problem is much reduced as only a small volume has "incorrect" air. It requires a flap valve in a third party component though, which is not exactly prudent.

3. This is smart. Pros: Doesn't need air supply, motor acts as valve and positive supply, and if we know motor rotation can give precise delivery profiles, and effectively a volume measurement. Cons: Less simple than solenoid valves, and no O2 enrichment

4. Also smart. Can probably be ranked w.r.t 3 according to number, cost of parts and complexity. How can we make these motors uber cheap/accessible?

5. Similar to 4, yes? But even simpler.

I think the 3-5 variants are better than 1, and I'd rate 2 the lowest. Ranking 3-5 should be able to be done using a matrix.

Questions:
 
 1. Can we use a simple 12v solenoid and a beam to do the see-saw shutoff of concept 4?

 2. Can we use 3 solenoid valves in place of the motor cam valve? Then we get 3!=6 settings, or 7 if one includes all closed.

hammockman, 2020-03-26:
I don't get Q2. explain?

hammockman, 2020-03-26:
1. is obvious and doable
2.

racleave, 2020-03-26:
After talking it seems like we should go with 1. One main reason is we don't have the flexible (probably latex) hose that we need for the hose valves.

Having said that, maybe we can design a rubber hose valve so it can be swapped out with our solenoid valves with 1" BSP female-female fittings.