hammockman/breather #13: Humidifier

racleave, 2020-03-27:
We probably need some humidity control.

From wikipedia: "According to that standard, the minimum water content of inspired respiratory gas is ca. 33 mg/dm³ and the maximum respiratory gas temperature is ca. 42 °C."

A bubble humidifier seems like the simplest solution.

 - We measure the humidity after the humidifier, and info is sent to the user by the computer.
 - The operator is responsible for changing water, including changing temperature to modify water entrainment
 - Can we have a simple control of the air coming into the humidifier - i.e. small bubbles instead of large?
 - Do we need a water trap after the dehumidifier?

hammockman, 2020-03-27:
Saturation @25 degC is only 23 mg/dm^3 and I'm pretty sure I'm breathing ok. Mind you, sometimes my mouth does get a bit dry... I shall investigate further.

racleave, 2020-03-28:
Maybe it is the warming you do in your nose. At 42 degrees (the max that standard gives) and 1021 hPa pressure (currently in Rotorua), that water content of 33 mg/dm3 is 60% relative humidity. That's quite reasonable, but drop the temp below 32 deg and we're at saturation and the lines are going to get soggy...