hammockman/breather #14: O2 sensor

racleave, 2020-03-27:
The O2 sensor I bought is 2-wire. I wish I'd bought the 3 or 4 wire version, as the 2-wire version doesn't have a heater and they need to come up to 300 deg C before they work.

Will look at pulling one out of a local car (Jon's, Kylie's, Steffan's, Jo's, in that order).

Until then we can use the 2-wire one as the mechanical template.

racleave, 2020-03-27:
If we can't get an O2 sensor working, we may be able to use the computer to:

 - measure temp and pressure of pure O2 line
 - measure temp and pressure of air o2 line
 - measure opening time of mixing valve
 - calculate O2 concentration

Even if this works we will still need to verify it is working. I woke up thinking about checking the colour and sound of the flame when mixed with a bit of LPG, but there must be some reason they say no smoking near these machines...

hammockman, 2020-03-27:
hot wire anemometers too?