hammockman/breather #21: Status 29/3

racleave, 2020-03-29:
A wee status for today:

 - We've got the basics of the prototype working. Here is a photo, and there are some videos floating around.

[image]

 - We have a display and control system that transmits over the local wifi network, so a web browser can display the readings and control settings (today's version below). Having multiple viewers is possible (phones, tablets, central PCs).

[image]

Sensor status (planned sensors):

Sensor | Physical | Electrical
--|--|--
Insp. temperature | Y | Y
Insp. pressure | Y | Y
Insp. flow | Y | N
O2 sensor | Y | N
Insp. humidity | N | N

To do:

 1. Improve on what is shown on "screen"
 1. Electrical connections of installed sensors
 1. Install exhaust valve = either 230v solenoid valve (not pilot operated we don't think) with a 12/230v relay, or the irrigation valve we modified this morning
 1. Install and connect humidity sensor
 1. Make and install bubble humidifier
 1. Install Oxygen line with valve. Use Argon for starters, then get an O2 bottle from Phil.

Notes:

 - It looks like a single pressure sensor in the inspiration line can also be used to measure and control BEEP. This is only if there is no flap valve at the Y-connection
 - It would be nice to have a control pressure measurment - not for the design but for the prototype setup and verification. This would ideally be placed near the patient

racleave, 2020-03-29:
We'll also need a box - approx l x w x h = 500 x 200 x 150 mm (the humidifier and inspiration/exhaust filters to be on outside of box). This is the setup as Jo left it today, with exhaust line in place but not connected.

[image]

hammockman, 2020-03-29:
I'm just about to check in an updated version of breathe and the UI. Breathe should now do a rudimentary breath cycle. The UI is pretty much complete, but ugly.